elegooofficial/ElegooSlicer #10: Gcode Layer slider Issue

Stickmasterdig, 2025-02-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### ElegooSlicer Version

1.1.7.3

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Neptune 3 pro

### How to reproduce

You have to upload a model and slice it and go to see the gecode code viewing and use the slider you see the gecode change.

[image]

### Actual results

[image] you see a gecode text text to model that cant be removed like in the image.

### Expected results

it should be fixed

### Project file & Debug log uploads

[New Compressed (zipped) Folder.zip](https://github.com/user-attachments/files/18834947/New.Compressed.zipped.Folder.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Stickmasterdig, 2025-02-18:
No I want to know how to remove the gecode displayed some there is not gecode

Stickmasterdig, 2025-02-18:
The command in the image

anjis-elegoo, 2025-02-18:
I don't quite understand what you mean. If you're looking to hide the GCode preview, you can follow the steps shown in the picture below.
[image]

closed 2025-02-18 by Stickmasterdig