elegooofficial/ElegooSlicer #12: For convenience there is also a portable build available - has been copy / pasted from Orca description,does not exist for ElegooSlicer?

sadsephiroth, 2025-02-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### ElegooSlicer Version

All

### Operating System (OS)

Windows

### OS Version

All

### Additional system information

Per https://github.com/ELEGOO-3D/ElegooSlicer#how-to-install

"How to install
Windows:

Download the installer for your preferred version from the [releases page](https://github.com/ELEGOO-3D/ElegooSlicer/releases).
For convenience there is also a portable build available."

Please correct or provide.

### Printer

Centauri Carbon

### How to reproduce

1. Read install instructions
2. Not listed under assets doe any release

### Actual results

No portable version available

### Expected results

Portable version available

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

anjis-elegoo, 2025-02-24:
Got it! Now, a portable version has been added for Windows.

sadsephiroth, 2025-02-24:
Awesome, thanks!

closed 2025-03-12 by anjis-elegoo