elegooofficial/ElegooSlicer #13: Linux build

russellmilliner, 2025-02-22:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

no

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I'd like a linux flatpak build of this slicer.  Is there intention to release a Linux build of this slicer?

wujie-elegoo, 2025-02-24:
It will be supported later, but it will take a while.

russellmilliner, 2025-02-24:
So I pre-ordered one with a "Before June 30, 2025" shipping.  I only have Linux.  Will I be able to use this printer or should I cancel my order?  There's no Linux version of the Elegoo software, and there's not yet Orca profiles.  Both are "coming soon", but promises and reality don't always match.

wujie-elegoo, 2025-02-24:
> So I pre-ordered one with a "Before June 30, 2025" shipping. I only have Linux. Will I be able to use this printer or should I cancel my order? There's no Linux version of the Elegoo software, and there's not yet Orca profiles. Both are "coming soon", but promises and reality don't always match.

The profiles for the new printer has been submitted to OrcaSlicer.  ElegooSlicer has just been launched, and the Linux version requires some development and testing manpower, which is currently still in the recruitment process.

peanutsguy, 2025-03-06:
I was able to compile for Linux (Ubuntu 22.04) a couple of weeks ago, but it wasn't loading the profiles. Haven't been able to dig into it any deeper.

russellmilliner, 2025-03-12:
You really should just put your efforts/money into making Orca better rather than forking/branding your own slicer.  That would be much more impressive than Elegoo Slicer.  Focus on what you do best and make good hardware then support good open software.

zeitue, 2025-03-12:
What's the ETA on Linux support for this? Also if this is really a fork of the OrcaSlicer which already has Linux support then it should be easy to just copy their build process.

ArKay74, 2025-07-26:
Would be nice if anything was working. But with Orca there seems to be an issue with rendering on my desktop. There's already a bug for that.

I tried running software in Wine, also got issues.

Tried it in a VM with Windows 11, that needs OpenGL 2. So yeah, I'm stuck. New printer, no software. Nice.

ArKay74, 2025-07-26:
> Use an Orca nightly build

Orca doesn't work either on my desktop, there seems to be a rendering issue (there's also a bug for that).

Couldn't get anything to work in Wine either. And in a Windows 11 VM I had no luck since it complains about missing OpenGL 2.

So yeah, new printer, no software to use it with. Nice.

russellmilliner, 2025-07-26:
> Orca doesn't work either on my desktop, there seems to be a rendering issue (there's also a bug for that).
>
> Couldn't get anything to work in Wine either. And in a Windows 11 VM I had no luck since [...]

did you try the flatpak?

russellmilliner, 2025-07-27:
> > did you try the flatpak?
> 
> Yes, that's what I downloaded.
> 
> After finding out that it's an issue with Wayland--since it works just fine when I log in with x11 instead--I eventually got it to work by running a Xephyr x11 server on a 2nd desktop and starting it on there. It's a good workaround until the bug is fixed I guess. At least I can use it now.

Just to note, I run Wayland (Manjaro) and use the flatpak with no issues.

ArKay74, 2025-07-27:
> Just to note, I run Wayland (Manjaro) and use the flatpak with no issues.

Probably not on an RTX 5090 I guess. Running Pop!_OS with Ubuntu 22.04.

Jonnyworks, 2025-08-07:
You can use ElegooSlicer's `BuildLinux.sh` to build for your linux distro from source.

They did use Docker in a creative/unsafe way as a build-server/engine for cross-distro support, you could use `DockerBuild.sh` to build and `DockerRun.sh` to run if `BuildLinux.sh` fails on your distro.

### BUT currently there's issues with the building process, causing build failure.

**I managed to get it running for myself** (on Gnome, Debian 12 Bookwork) and I'll make a fork/pull-request in the coming days trying to aim for flatpak support. **Here's the current problems**:

1. Outdated dependency URLs/SHA256 for `dep/TIFF/TIFF.cmake` and `dep/MPFR/MPFR.cmake`
2. Syntax errors in `localization/i18n/fr/ElegooSlicer_fr.po`
3. Deprecated AppImage Tool URL in `src/platform/unix/build_appimage.sh.in`

I tried OrcaSlicer myself, but didn't get good results yet and I didn't want to finetune everything myself. It currently lacks profiles for Neptune 4 printers and Elegoo filaments.

Jonnyworks, 2025-08-07:
Thank you! That's exactly how I fixed it, too. I decided to stick with the .AppImage, since it works just fine. No Flatpak required.

ArKay74, 2025-08-07:
[image]

Jonnyworks, 2025-08-07:
> Tried it in distrobox with ubuntu:latest, also no luck.
> 
> ```
> Creating Appimage for distribution...~/Downloads/ElegooSlicer/build/package ~/Downloads/ElegooSlicer/build
> --2025-08-07 17:36:59--  https://github.com/AppImage/appimagetool/releases/download/continuous/appimagetool-x86_64.AppImage
> Resolving github.com (github.com)... 140.82.121.4
> Connecting to github.com (github.com)|140.82.121.4|:443... connected.
> HTTP request sent, awaiting response... 302 Found
> Location: https://release-assets.githubusercontent.com/github-production-release-asset/658450330/a7b27126-c6bc-47bc-b13b-eae1a0d3419d?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-08-07T16%3A10%3A15Z&rscd=attachment%3B+filename%3Dappimagetool-x86_64.AppImage&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-08-07T15%3A09%3A53Z&ske=2025-08-07T16%3A10%3A15Z&sks=b&skv=2018-11-09&sig=TZGwKEzV0keNcdUXiQYA0xacNIaS%2BnVlgmaKhsxknms%3D&jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1NDU4MTMxOSwibmJmIjoxNzU0NTgxMDE5LCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.pLI-uaCxeMmL3mj1l-KlU3V_AZ87GXOYGCl9gOOT3X4&response-content-disposition=attachment%3B%20filename%3Dappimagetool-x86_64.AppImage&response-content-type=application%2Foctet-stream [following]
> --2025-08-07 17:36:59--  https://release-assets.githubusercontent.com/github-production-release-asset/658450330/a7b27126-c6bc-47bc-b13b-eae1a0d3419d?sp=r&sv=2018-11-09&sr=b&spr=https&se=2025-08-07T16%3A10%3A15Z&rscd=attachment%3B+filename%3Dappimagetool-x86_64.AppImage&rsct=application%2Foctet-stream&skoid=96c2d410-5711-43a1-aedd-ab1947aa7ab0&sktid=398a6654-997b-47e9-b12b-9515b896b4de&skt=2025-08-07T15%3A09%3A53Z&ske=2025-08-07T16%3A10%3A15Z&sks=b&skv=2018-11-09&sig=TZGwKEzV0keNcdUXiQYA0xacNIaS%2BnVlgmaKhsxknms%3D&jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmVsZWFzZS1hc3NldHMuZ2l0aHVidXNlcmNvbnRlbnQuY29tIiwia2V5Ijoia2V5MSIsImV4cCI6MTc1NDU4MTMxOSwibmJmIjoxNzU0NTgxMDE5LCJwYXRoIjoicmVsZWFzZWFzc2V0cHJvZHVjdGlvbi5ibG9iLmNvcmUud2luZG93cy5uZXQifQ.pLI-uaCxeMmL3mj1l-KlU3V_AZ87GXOYGCl9gOOT3X4&response-content-disposition=attachment%3B%20filename%3Dappimagetool-x86_64.AppImage&response-content-type=application%2Foctet-stream
> Resolving release-assets.githubusercontent.com (release-assets.githubusercontent.com)... 185.199.111.133, 185.199.109.133, 185.199.110.133, ...
> Connecting to release-assets.githubusercontent.com (release-assets.githubusercontent.com)|185.199.111.133|:443... connected.
> HTTP request sent, awaiting response... 200 OK
> Length: 15112696 (14M) [application/octet-stream]
> Saving to: '../appimagetool.AppImage'
> 
> ../appimagetool.AppImage               100%[=========================================================================>]  14.41M  --.-KB/s    in 0.1s    
> 
> 2025-08-07 17:37:00 (98.6 MB/s) - '../appimagetool.AppImage' saved [15112696/15112696]
> 
> ../build_appimage.sh: 35: ../appimagetool.AppImage: not found
> mv: cannot stat 'ElegooSlicer-x86_64.AppImage': No such file or directory
> chmod: cannot access 'ElegooSlicer_Linux_V1.1.8.0.AppImage': No such file or directory
> ~/Downloads/ElegooSlicer/build
> mv: cannot stat 'package/ElegooSlicer_Linux_V1.1.8.0.AppImage': No such file or directory
> done
> ~/Downloads/ElegooSlicer
> done
> 📦[arkay@ubuntu-latest ElegooSlicer]> 
> ```

Your output is related to the .AppImage and can't seem to find the appimagetool(?), maybe permission issues?.

If the build succeeded, you should at least see the application under `./build/package/bin/elegoo-slicer` and start it. I built it with `./BuildLinux.sh -dsir`. If you share the entire log, I could tell you more possibly.

Jonnyworks, 2025-08-07:
> > Your output is related to the .AppImage and can't seem to find the appimagetool(?), maybe permission issues?.
> > If the build succeeded, you should at least see the application under `./build/package/bin/elegoo-slicer` and start it. I built it with `./BuildLinux.sh -dsir`. If you share the entire log, I could tell you more possibly.
> 
> Yeah, I think this script has a couple of path issues. I gave up on it. The application it built was core-dumping. Running it in distrobox is probably also not the best solution.

You could try starting it with ./build/package/bin/AppRun

It mentions some workaround to avoid segfaults for arch based distros.

chan32167, 2025-08-15:
After a couple of hours was able to compile it in ubuntu 24.04 

[image]

in case anyone wants the appimage i uploaded it here https://drive.google.com/file/d/170AslSRo_ynlG8p_57JjtgPU4MGQ-COe/view?usp=drive_link

Jonnyworks, 2025-11-14:
> > I know, not supported, etc. but if I can fix this, I think I'll get a good appimage.
> 
> Their site says:
> 
> "Download Elegoo Slicer Version: Elegoo Slicer v1.1.8.2 Release date: 12 March 2025 Licence: MIT & Apache‑2.0 — open source Minimum OS: Win 10/11 ∙ macOS 12+ ∙ **Ubuntu 22.04+** Source & changelog: GitHub release page -- https://elegooslicer.com/#download"
> 
> Feel like I should have read the issues here before buying. I'm hours in in trying to compile.
> 
> CMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message): Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS) (Required is at least version "3.1") Call Stack (most recent call first): /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE) /usr/share/cmake-3.28/Modules/FindwxWidgets.cmake:988 (find_package_handle_standard_args) src/CMakeLists.txt:49 (find_package)
> 
> The errors are endless...

Yeah, last time I build it for debian (also X11), there were multiple issues as well (see my initial comment above), you're probably struggling with the same issues for different packages.

While looking through their repo, I also found they support Docker which hacks in a X11 display env. That means you could try running `DockerBuild.sh` followed by `DockerRun.sh` and hopefully avoid this dependency hell. Of course, you need docker installed.

From my experience, this repo is basically almost dead and no linux support is being actively worked on.

FilleMang, 2025-11-14:
I just gave up, sliced in Orca, started my first print. And yes, your .AppImage worked perfect. I spent all Friday night trying to resolve dependencies in all the worst ways for not...
It seems you have given us all a workaround. Your instruction will save a lot of frustration for someone else. Thank you!

zeitue, 2025-11-18:
Since it can be built in Docker and yet no guarantee of running it on a Linux host, why not a flatpak, which can target a specific runtime?

hnnrs1092, 2025-11-20:
I seemed to have no problems compiling on a fresh install of ubuntu 24.04 just simply using sudo ./BuildLinux.sh -u then ./BuildLinux.sh -dsir. The AppImage built first time. I had also compiled in 22.04 on my main linux box but did run into some issues with dep files but eventually got a working AppImage which I currently use. However, I tried to recompile recently and failed again. I have to look into this again but I thought I should let others know it was successful on 22.04.

DarmainTheDonkey, 2026-01-02:
Hi.  Got myself a Centauri Carbon.  All my compute platforms are Linux, Ubuntu 24.04.  So replicated a workspace and tried to build the project for Linux.  I first checked out the latest TAG but that seems to be broken, so I went to the tip of MAIN.  I can't build this project. Here's what I get (After breaking down the steps to find the root cause) -
./BuildLinux.sh -c - OK
./BuildLinux.sh -d - OK
./BuildLinux.sh -s - Machine gradually slows and appears locked up.  I tried again, but this time with nothing else running, except another shell session with HTOP running.  This showed the build process had consumed all available RAM and was caning the swap space.  I have 24 Gb of RAM.  I also have 24 cores, so I reasoned to try a ./BuildLinux.sh -1 to limit the number of threads.  This made no difference, it still ran out of RAM.
I've got an Enterprise Server, with 48Gb of RAM, so I am thinking to fire that up and try the build on that.
What kind of spec machines are you guys running?  Any suggestions on how to get this under control?
Many thanks.

TomT15, 2026-01-03:
i am not sure what to make of this

make[1]: Leaving directory '/home/{username}/app/ElegooSlicer-1.1.8.2/deps/build/dep_OpenSSL-prefix/src/dep_OpenSSL'
ninja: build stopped: subcommand failed.

more info via grep

```
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/templates/OpenCVConfig.root-WIN32.cmake.in: No such file or directory
error: deps/build/dep_OpenCV-prefix/src/dep_OpenCV/cmake/OpenCVDetectCXXCompiler.cmake: No such file or directory
-- Installing: /home/{username}/app/ElegooSlicer-1.1.8.2/deps/build/destdir/usr/local/include/jerror.h
```

Edit: syntax and grammer

TomT15, 2026-01-04:
I can confirm I DID NOT use `gh repo clone` and instead opted for direct download of the file. I will start there when i have a chance as well

ruben0909, 2026-01-07:
Good morning,

I hope I’m asking in the right place. If not, where should I direct my question?

Is there an official release date for pre-compiled Linux builds?

Thank you very much, best regards.

MikaelElkiaer, 2026-02-04:
> From my perspective, I would leave this repo behind and head back to [orcaslicer](https://github.com/OrcaSlicer/OrcaSlicer). There you can use your printer without hassle and it has linux builds. Nothing happend in this repo (which is some kind of fork of orcaslicer) since March 2025 (the fixes in August for Linux apparently were not successful).

How do you get an appropriate printer (and possibly filament) profile?

NovaViper, 2026-04-22:
The appimage doesn't work on NixOS sadly.. keep getting this error
```
ElegooSlicer_Linux_V1.5.0.7.appimage installed in /home/novaviper/.cache/appimage-run/f588eec0fd8e0064457ea78abf291101d48fd2a977d7522e8b7a81adc00e9b6d
/home/novaviper/.cache/appimage-run/f588eec0fd8e0064457ea78abf291101d48fd2a977d7522e8b7a81adc00e9b6d/bin/elegoo-slicer: error while loading shared libraries: libbz2.so.1.0: cannot open shared object file: No such file or directory
```

HattonLe, 2026-10-08:
I don't have a Elegoo printer as yet, so was just checking out the software side of things first for usability on linux and found this thread.
Heres an update as of right now using my system, Ubuntu 26.04 kernel 7.0... 3.8GHz, 8 threads, 24GB RAM.
From a previous comment, found the linux V1.5.3.5.AppImage at https://github.com/elegooofficial/ElegooSlicer/releases
Not used that particular 3D printer app before but I can load an STL in, and slice it, view and rotate it etc.., so that seems to run ok.
From a previous comment on building sources, I did the following
git clone --recursive https://github.com/ELEGOO-3D/ElegooSlicer
cd ElegooSlicer
./build_linux.sh -u       <--- note this installed 45 packages.
./build_linux.sh -dsi

After about 15 mins I guess, build eventually failed at [207/235] Performing build step for 'dep_elegoolink'
/home/main/Downloads/ElegooSlicer/deps/build/dep_elegoolink-prefix/src/dep_elegoolink/include/types/../events/event_system.h:94:28: error: ‘find_if’ is not a member of ‘std’ [-Wtemplate-body]
   94 |             auto it = std::find_if(eventHandlers.begin(), eventHandlers.end(),