elegooofficial/ElegooSlicer #22: .3mf support? Orca slicer supports this fine?

Shybella, 2025-03-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Please add .3mf support.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Support .3mf

anjisZhou, 2025-03-27:
ElegooSlicer supports importing and exporting 3MF files. Could you describe the issue in more detail?

anjis-elegoo, 2025-03-27:
ElegooSlicer supports importing and exporting 3MF files. Could you describe the issue in more detail?

anjis-elegoo, 2025-03-28:
Is this 3MF file exported from Prusa? 3MF files exported from Prusa are not supported.

Shybella, 2025-03-28:
It's from Cura and Orca slicer supports it fine.

Elegooslicer is based off Orca. Prusa 3mf works fine with Orca.

anjis-elegoo, 2025-03-28:
Could you share the 3MF file? We'd like to take a look and figure out the cause.

cgolchert, 2025-09-17:
I'm getting errors about versions:

> Elegoo Slicer - Newer version
> 
> The 3mf's version 2.3.0.0 is newer than ElegooSlicer's version 2.2.0, Found the following keys unrecognized:
> 
> You should update your software.

This was from a 3df file of a CC riser file.