elegooofficial/ElegooSlicer #25: When starting to print, printer head immediately crashes into x axis

Tylersuard, 2025-04-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### ElegooSlicer Version

1.1.8.2

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

I am using an Elegoo OrangeStorm Giga.  I try to print a file created using your slicer.  After homing, the print head immediately crashes into the gantry along the x-axis.  I am not sure what is going on here.

### Printer

Elegoo OrangeStorm Giga

### How to reproduce

Try to slice and then export a file

### Actual results

No printing happens, the printer head just slams into the x axis gantry after homing

### Expected results

I expected it to print

### Project file & Debug log uploads

Here is the profile:
[print problems.zip](https://github.com/user-attachments/files/19934018/print.problems.zip)
Here is the gcode:

[EOGiga1_1_15_to_45_degree_overhang_test_PLA0.4_8m46s.zip](https://github.com/user-attachments/files/19934041/EOGiga1_1_15_to_45_degree_overhang_test_PLA0.4_8m46s.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

anjis-elegoo, 2025-05-07:
This issue might not be related to slicing — it could be a hardware problem. You can contact customer support at `3dp@elegoo.com` to report it, and a specialist will get in touch with you.

Tylersuard, 2025-05-17:
As it turns out the recommended dimensions for the Elegoo OrangeStorm are incorrect, changing them back to 800x800x1000 fixed this issue.

closed 2025-05-17 by Tylersuard