elegooofficial/ElegooSlicer #26: Invalid Object Collision Warning

LexxM3, 2025-05-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### ElegooSlicer Version

1.1.8.2

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro

### Printer

Centauri Carbon

### How to reproduce

See attached .3mf file and screen cap.  I suspect the issue can be reproduced whenever any object overall bounding box overlaps another, regardless of whether the actual objects overlap (i.e. bounding box is an invalid (or at least insufficient) method of detecting overlaps).

### Actual results

Slicer generates an invalid object overlap warning.  Slices and prints without issue, however.

[image]

[2025-05-04 Elegoo Slicer Bug, Invalid Overlap Warning.zip](https://github.com/user-attachments/files/20030178/2025-05-04.Elegoo.Slicer.Bug.Invalid.Overlap.Warning.zip)

### Expected results

No invalid overlap warnings.

### Project file & Debug log uploads

[2025-05-04 Elegoo Slicer Bug, Invalid Overlap Warning.zip](https://github.com/user-attachments/files/20030178/2025-05-04.Elegoo.Slicer.Bug.Invalid.Overlap.Warning.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

anjis-elegoo, 2025-05-07:
Currently, the boundary detection is based on the bounding box of the model. This may trigger a warning. Since the center of your model is hollow, there won’t be any actual printing issues. However, if the model were solid and placed like this, collisions could occur. You can also follow the steps shown in the image to combine the parts, which will prevent this issue.

[image]

LexxM3, 2025-05-07:
Yes, understood. It's still a bug, however ... one that doesn't, incidentally, exist in current or any recent releases of Bambu Studio. I just tested and it does, however, also exists in Orca, so you inherited it from there. You're welcome to forward the bug their way.