elegooofficial/ElegooSlicer #28: M190 in start gcode causing Centauri Carbon extremely long wait time when using cool plate (B side)

duke8253, 2025-05-19:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Since the bed levling is done at 60C, and the cool plate or B side only needs 30-35C, the M190 causes the printer to wait for a very long time (over 30 minutes) for the bed to cool down.

https://github.com/ELEGOO-3D/ElegooSlicer/blob/1abe50f97cc0e826e5e988106963b8dc4d159c79/resources/profiles/Elegoo/machine/ECC/Elegoo%20Centauri%20Carbon%200.4%20nozzle.json#L54

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

My suggestion is to change the bed levling process to use whatever the initial layer bed temperature of the print, not a static 60C. But that'll have to be done in the firmware.

wujie-elegoo, 2025-05-19:
Different heating and waiting strategies are needed for the B-side, which will be optimized soon. Thank you for your feedback.

Alexanderdubz, 2025-08-12:
I was planning on asking the same thing. I have tried multiple different additions and changes in the GCode, but still to no avail. Please, please, please make this extremely useful (and effectively necessary) change in the slicer <3

Alexflu, 2025-09-09:
@Lovelace3314 

G-code M8803 exports 3 configurations from the printer to the USB, if inserted directly after BED_MESH_CALIBRATE you can pull the bed mesh from that calibration. [besh_profile_enhancement_default] is the smooth plate Besh that is modified during this calibration. 

BED_MESH_CALIBRATE
M8807

I've not tried yet but it may be possible to insert the desired Besh profile and upload it with the M8807 G-code to have it save permanently, **proceed with extreme caution** here as errors could cause printer not to boot and essentially "brick" it.. which is why I haven't tried. I've been seeking another workaround as leveling per print when trying to tweak settings has been quite time consuming, and z-offset adjustments for a 60c mesh while printing at other bed temps at a .1mm layer height just isn't possible due to the bed tendency to curve..

Lovelace3314, 2025-09-10:
> [@Lovelace3314](https://github.com/Lovelace3314)
> 
> G 代码 M8803 将 3 个配置从打印机导出到 USB，如果在 BED_MESH_CALIBRATE 之后直接插入，则可以从该校准中提取床网格。user_printer.cfg 中的 [besh_profile_enhancement_default] 是在此校准期间修改的光滑板 Besh。
> 
> 床面网格校准 M8803
> 
> 我还没有尝试过，但也许可以编辑 [auto_leveling] bed_mesh_temp : 60.0 ，但这会编辑所有床网格，或者可以插入所需的 Besh 配置文件并将其与 M8807 G 代码一起上传以永久保存，在这里**要非常小心**，因为错误可能会导致打印机无法启动..这就是为什么我没有尝试修改配置

Thank you for your answer. The above operation is to replace the configuration file. I'm also very worried that the printer won't start. I still choose to wait and see for a while longer, and try again after others have fully verified it.