elegooofficial/ElegooSlicer #29: Per object flow ratio defaults to 0.97

WillJohns0n, 2025-07-16:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

In the default profiles for the Centauri Carbon. The flow ratio in the object settings defaults to 0.97, this causes under-extrusion as flow rate calibration is done assuming the per object flow rate will be set to 1. This means that the user has to remember to manually change this value for each job.

To recreate: Import a model to a new project using default profiles. Change the process settings from Global to Objects. Select the Quality tab and view the setting for Flow Ratio

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would be easier if the flow ratio is set to 1 as default

wujie-elegoo, 2025-07-28:
We will test the effect of setting the flow ratio to 1. Thanks for the suggestion.

hansonap, 2026-09-18:
Is there an update on this issue? Setting the default object value to .97 nullifies the YOLO Flow Rate results as designed.

Note: There is no way to set this value at the global Process level, so if the value needs to be other than .97, it impacts all objects of all prints.